syssi/esphome-atorch-dl24 #10: get CRC check failed

nhnam100, 2022-01-30:
I use the DL24  and get continue messages
ESPhome [W][atorch dl24:100]: CRC check failed. Skipping
The ESPhome is lasted version v2022.1.2
the information during compiling

INFO Reading configuration /config/esphome/ble_tracker.yaml...
INFO Updating https://github.com/syssi/esphome-atorch-dl24.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing bletracker (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ArduinoJson> 6.18.5
|-- <ESPmDNS> 1.0
...
so could not get the data;

syssi, 2022-01-30:
Could you check the firmware version of your DL24? I would like to compare it with my version.

nhnam100, 2022-01-30:
my DL24 menu shown V5.1

syssi, 2022-01-30:
I just checked my DL24. It's the same version (V5.1). This is my configuration using the latest esphome release (`2022.1.2`) without trouble:

```
substitutions:
  name: atorch-dl24

esphome:
  name: ${name}

esp32:
  board: esp32doit-devkit-v1

external_components:
  - source: external_components/esphome-atorch-dl24/components

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
api:

logger:
  level: DEBUG
  baud_rate: 0

sensor:
  - platform: atorch_dl24
    ble_client_id: dummyload0
    voltage:
      name: "${name} voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    capacity:
      name: "${name} capacity"
    energy:
      name: "${name} energy"
    temperature:
      name: "${name} temperature"
    dim_backlight:
      name: "${name} dim backlight"
    running:
      name: "${name} running"

esp32_ble_tracker:

ble_client:
  - mac_address: 41:B8:12:0A:A8:37
    id: dummyload0
    on_connect:
      then:
        - lambda: |-
            ESP_LOGD("ble_client_lambda", "Connected to BLE device");
    on_disconnect:
      then:
        - lambda: |-
            ESP_LOGD("ble_client_lambda", "Disconnected from BLE device");
```

This is my log output:

```
INFO Successfully connected to atorch-dl24.local
[16:27:26][D][ble_client:047]: Found device at MAC address [41:B8:12:0A:A8:37]
[16:27:26][I][ble_client:085]: Attempting BLE connection to 41:b8:12:0a:a8:37
[16:27:27][D][ble_client_lambda:065]: Connected to BLE device
[16:27:27][I][ble_client:161]: Service UUID: 0x1800
[16:27:27][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x9
[... 40 lines cut ...]
[16:27:32][D][sensor:125]: 'atorch-dl24 voltage': Sending state 0.00000 V with 1 decimals of accuracy
[16:27:32][D][sensor:125]: 'atorch-dl24 current': Sending state 0.00000 A with 3 decimals of accuracy
[16:27:32][D][sensor:125]: 'atorch-dl24 power': Sending state 0.00000 W with 4 decimals of accuracy
[16:27:32][D][sensor:125]: 'atorch-dl24 capacity': Sending state 88.02000 Ah with 4 decimals of accuracy
[16:27:32][D][sensor:125]: 'atorch-dl24 energy': Sending state 280.00000 Wh with 0 decimals of accuracy
[16:27:32][D][sensor:125]: 'atorch-dl24 temperature': Sending state 19.00000 °C with 0 decimals of accuracy
[16:27:32][D][atorch_dl24:161]:   Timer: 04:24:05
[16:27:32][D][sensor:125]: 'atorch-dl24 running': Sending state 0.00000  with 0 decimals of accuracy
[16:27:32][D][sensor:125]: 'atorch-dl24 dim backlight': Sending state 60.00000 s with 0 decimals of accuracy
[16:27:33][D][sensor:125]: 'atorch-dl24 voltage': Sending state 0.00000 V with 1 decimals of accuracy
[16:27:33][D][sensor:125]: 'atorch-dl24 current': Sending state 0.00000 A with 3 decimals of accuracy
[16:27:33][D][sensor:125]: 'atorch-dl24 power': Sending state 0.00000 W with 4 decimals of accuracy
[16:27:33][D][sensor:125]: 'atorch-dl24 capacity': Sending state 88.02000 Ah with 4 decimals of accuracy
[16:27:33][D][sensor:125]: 'atorch-dl24 energy': Sending state 280.00000 Wh with 0 decimals of accuracy
[16:27:33][D][sensor:125]: 'atorch-dl24 temperature': Sending state 19.00000 °C with 0 decimals of accuracy
[16:27:34][D][sensor:125]: 'atorch-dl24 voltage': Sending state 0.00000 V with 1 decimals of accuracy
[16:27:34][D][sensor:125]: 'atorch-dl24 current': Sending state 0.00000 A with 3 decimals of accuracy
[16:27:34][D][sensor:125]: 'atorch-dl24 power': Sending state 0.00000 W with 4 decimals of accuracy
[16:27:34][D][sensor:125]: 'atorch-dl24 capacity': Sending state 88.02000 Ah with 4 decimals of accuracy
[16:27:34][D][sensor:125]: 'atorch-dl24 energy': Sending state 280.00000 Wh with 0 decimals of accuracy
[16:27:34][D][sensor:125]: 'atorch-dl24 temperature': Sending state 19.00000 °C with 0 decimals of accuracy
[16:27:34][D][atorch_dl24:161]:   Timer: 04:24:05
```

Could you provide your configuration for comparison? What's the distance between your DL24 and the ESP32?

nhnam100, 2022-01-31:
my code
----
substitutions:
  _device_name: "bletracker"
  _ssid: "xxxx"
  _passwifi: "xxxx"
  _ssid1: "xxxx"
  _passwifi1: "xxxx"
  _passota: "xxxx"
  _passapi: "xxxxx"
   _passap: "xxxxx"
  _manip: 192.168.1.113
  _mangw: 192.168.1.1
  _broker: 192.168.1.130
  _usernameMQTT: "xxxx"
  _passwordMQTT: "xxxx"
  
esphome:
  name: $_device_name
  platform: ESP32
  board: nodemcu-32s

wifi:
  networks:
  - ssid: $_ssid
    password: $_passwifi
  - ssid: $_ssid1
    password: $_passwifi1

  manual_ip:
    static_ip: $_manip
    gateway: $_mangw
    subnet: 255.255.255.0

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $_device_name Fallback Hotspot
    password: $_passap 
  
captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: $_passapi 

ota:
  password: $_passota

# Example configuration entry
web_server:
  port: 80 

external_components:
  - source: github://syssi/esphome-atorch-dl24@main
    refresh: 0s

esp32_ble_tracker:

#mqtt:
#  broker: $_broker
#  username: $_usernameMQTT
#  password: $_passwordMQTT
#  id: mqtt_client

ble_client:
  - mac_address: B0:00:00:08:AE:CF
    id: dummyload0
[... 106 lines cut ...]
[18:04:27][C][ota:086]:   Using Password.
[18:04:27][C][api:134]: API Server:
[18:04:27][C][api:135]:   Address: 192.168.1.113:6053
[18:04:27][C][api:139]:   Using noise encryption: NO
[18:04:27][C][wifi_signal.sensor:009]: WiFi Signal 'bletracker WiFi Signal'
[18:04:27][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[18:04:27][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[18:04:27][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[18:04:27][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[18:04:27][C][mdns:084]: mDNS:
[18:04:27][C][mdns:085]:   Hostname: bletracker
[18:04:28][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:28][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:29][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:29][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:30][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:30][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:31][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:31][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:32][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:32][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:33][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:33][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:34][W][atorch_dl24:100]: CRC check failed. Skipping frame.
[18:04:34][W][atorch_dl24:100]: CRC check failed. Skipping frame.

nhnam100, 2022-02-03:
during compiling, there is a message but still generated the bin files 

Compiling /data/ble_tracker/.pioenvs/ble_tracker/FrameworkArduino/esp32-hal-time.c.o
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

syssi, 2022-02-03:
Could you increase the log level to very verbose?

```
logger:
  level: VERY_VERBOSE
```

You should see the received raw data produced by https://github.com/syssi/esphome-atorch-dl24/blob/main/components/atorch_dl24/atorch_dl24.cpp#L78

Could you provide this data/logs?

nhnam100, 2022-02-23:
Sorry for long respond
My log file as you proposed
[logs_atorch-ble_upload (1).txt](https://github.com/syssi/esphome-atorch-dl24/files/8121634/logs_atorch-ble_upload.1.txt)
Thanks

syssi, 2022-02-23:
```
# Log of a working device
[09:17:15][VV][esp32_ble_tracker:497]: Parse Result:
[09:17:15][VV][esp32_ble_tracker:514]:   Address: 41:B8:12:0A:A8:37 (PUBLIC)
[09:17:15][VV][esp32_ble_tracker:516]:   RSSI: -77
[09:17:15][VV][esp32_ble_tracker:517]:   Name: 'DL24_BLE'
[09:17:15][VV][esp32_ble_tracker:525]:   Ad Flag: 6
[09:17:15][VV][esp32_ble_tracker:528]:   Service UUID: 0xFFE0
[09:17:15][VV][esp32_ble_tracker:547]: Adv data: 02.01.06.03.03.E0.FF.1E.09.44.4C.32.34.5F.42.4C.45.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (38)
[09:17:15][D][ble_client:047]: Found device at MAC address [41:B8:12:0A:A8:37]
[09:17:15][I][ble_client:085]: Attempting BLE connection to 41:b8:12:0a:a8:37
[09:17:16][V][ble_client:115]: [41:b8:12:0a:a8:37] ESP_GATTC_OPEN_EVT
[09:17:16][D][ble_client_lambda:065]: Connected to BLE device
[09:17:17][V][ble_client:134]: cfg_mtu status 0, mtu 248
[09:17:17][V][ble_client:159]: [41:b8:12:0a:a8:37] ESP_GATTC_SEARCH_CMPL_EVT
[09:17:17][I][ble_client:161]: Service UUID: 0x1800
[09:17:17][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x9
[09:17:17][I][ble_client:347]:  characteristic 0x2A00, handle 0x3, properties 0x2
[09:17:17][I][ble_client:347]:  characteristic 0x2A01, handle 0x5, properties 0x2
[09:17:17][I][ble_client:347]:  characteristic 0x2A02, handle 0x7, properties 0x2
[09:17:17][I][ble_client:347]:  characteristic 0x2A04, handle 0x9, properties 0x2
[09:17:17][I][ble_client:161]: Service UUID: 0xFFE0
[09:17:17][I][ble_client:162]:   start_handle: 0xa  end_handle: 0xf
[09:17:17][I][ble_client:347]:  characteristic 0xFFE1, handle 0xc, properties 0x1c
[09:17:17][V][ble_client:382]:    descriptor 0x2902, handle 0xd
[09:17:17][I][ble_client:347]:  characteristic 0xFFE2, handle 0xf, properties 0xc
[09:17:17][D][esp32_ble_tracker:217]: Starting scan...
[09:17:17][VV][scheduler:026]: set_timeout(name='scan', timeout=600000)
[09:17:18][VV][atorch_dl24:078]: Notification received: FF.55.01.02.00.00.00.00.00.00.00.22.62.00.00.00.1C.00.00.00.00.00.00.00.00.13.00.04.18.05.3C.00.00.00.00.57 (36)
[09:17:18][D][sensor:125]: 'basement-btle-gateway voltage': Sending state 0.00000 V with 1 decimals of accuracy
[09:17:18][D][sensor:125]: 'basement-btle-gateway current': Sending state 0.00000 A with 3 decimals of accuracy
[09:17:18][D][sensor:125]: 'basement-btle-gateway power': Sending state 0.00000 W with 4 decimals of accuracy
[09:17:18][D][sensor:125]: 'basement-btle-gateway capacity': Sending state 88.02000 Ah with 4 decimals of accuracy
[09:17:18][D][sensor:125]: 'basement-btle-gateway energy': Sending state 280.00000 Wh with 0 decimals of accuracy
[09:17:18][D][sensor:125]: 'basement-btle-gateway temperature': Sending state 19.00000 °C with 0 decimals of accuracy
[09:17:18][D][atorch_dl24:161]:   Timer: 04:24:05
[09:17:18][D][sensor:125]: 'basement-btle-gateway dim backlight': Sending state 60.00000 s with 0 decimals of accuracy
```

```
# Your device
[10:04:34][VV][esp32_ble_tracker:497]: Parse Result:
[10:04:34][VV][esp32_ble_tracker:514]:   Address: B0:00:00:08:AE:CF (PUBLIC)
[10:04:34][VV][esp32_ble_tracker:516]:   RSSI: -83
[10:04:34][VV][esp32_ble_tracker:517]:   Name: 'DL24-BLE'
[10:04:34][VV][esp32_ble_tracker:525]:   Ad Flag: 6
[10:04:34][VV][esp32_ble_tracker:528]:   Service UUID: 0xFFE0
[10:04:34][VV][esp32_ble_tracker:528]:   Service UUID: 0xFEE7
[10:04:34][VV][esp32_ble_tracker:531]:   Manufacturer data: 88.A0.B0.00.00.07.AE.CF (8)
[10:04:34][VV][esp32_ble_tracker:547]: Adv data: 02.01.06.05.02.E0.FF.E7.FE.0B.FF.C1.8D.88.A0.B0.00.00.07.AE.CF.09.09.44.4C.32.34.2D.42.4C.45 (31)
[10:04:34][D][ble_client:047]: Found device at MAC address [B0:00:00:08:AE:CF]
[10:04:35][V][ble_client:159]: [b0:00:00:08:ae:cf] ESP_GATTC_SEARCH_CMPL_EVT
[10:04:35][I][ble_client:161]: Service UUID: 0x1800
[10:04:35][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x9
[10:04:35][I][ble_client:347]:  characteristic 0x2A00, handle 0x3, properties 0x2
[10:04:35][I][ble_client:347]:  characteristic 0x2A01, handle 0x5, properties 0x2
[10:04:35][I][ble_client:347]:  characteristic 0x2A02, handle 0x7, properties 0x2
[10:04:35][I][ble_client:347]:  characteristic 0x2A04, handle 0x9, properties 0x2
[10:04:35][I][ble_client:161]: Service UUID: 0x1801
[10:04:35][I][ble_client:162]:   start_handle: 0xa  end_handle: 0xd
[10:04:35][I][ble_client:347]:  characteristic 0x2A05, handle 0xc, properties 0x20
[10:04:35][V][ble_client:382]:    descriptor 0x2902, handle 0xd
[10:04:35][I][ble_client:161]: Service UUID: 0xFFE0
[10:04:35][I][ble_client:162]:   start_handle: 0xe  end_handle: 0xffff
[10:04:35][I][ble_client:347]:  characteristic 0xFFE1, handle 0x10, properties 0x1c
[10:04:35][V][ble_client:382]:    descriptor 0x2902, handle 0x11
[10:04:35][I][ble_client:347]:  characteristic 0xFFE2, handle 0x13, properties 0xc
[10:04:35][VV][atorch_dl24:078]: Notification received: FF.55.01.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (20)
[10:04:35][W][atorch_dl24:105]: Frame skipped because of invalid length. USB meter report frames aren't supported right now.
```

syssi, 2022-02-23:
It looks like your hardware is different. F.e. your BLE module provides some manufacturer data:

```
[10:04:34][VV][esp32_ble_tracker:531]:   Manufacturer data: 88.A0.B0.00.00.07.AE.CF (8)
```

and a additional service UUID:

```
[10:04:34][VV][esp32_ble_tracker:528]:   Service UUID: 0xFEE7
```

The advertised data is shorter (31 bytes vs. 38 bytes):

```
[10:04:34][VV][esp32_ble_tracker:547]: Adv data: 02.01.06.05.02.E0.FF.E7.FE.0B.FF.C1.8D.88.A0.B0.00.00.07.AE.CF.09.09.44.4C.32.34.2D.42.4C.45 (31)
[09:17:15][VV][esp32_ble_tracker:547]: Adv data: 02.01.06.03.03.E0.FF.1E.09.44.4C.32.34.5F.42.4C.45.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (38)
```

And if the BLE client subscribes to the service UUID `0xFFE0` + characteristics `0xFFE1` the notification data is always divided into two chunks and 36 bytes long in total:

``` 
[10:04:38][VV][atorch_dl24:078]: Notification received: FF.55.01.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (20)
[10:04:38][VV][atorch_dl24:078]: Notification received: 00.00.00.00.00.1E.00.00.00.00.3C.00.00.00.00.19 (16)
```

vs.  single notification response (36 bytes)

```
[09:17:18][VV][atorch_dl24:078]: Notification received: FF.55.01.02.00.00.00.00.00.00.00.22.62.00.00.00.1C.00.00.00.00.00.00.00.00.13.00.04.18.05.3C.00.00.00.00.57 (36)
```

I assume we have to combine the two notifications into one.

syssi, 2022-02-23:
The CRC of the combined message is fine:

```
> const packet = Buffer.from('FF5501020000000000000000000000000000000000000000001E000000003C0000000019', 'hex');
> const payload = packet.slice(2, -1);
> const checksum = payload.reduce((acc, item) => (acc + item) & 0xff, 0) ^ 0x44;
> packet[packet.length - 1] == checksum;
true
```

syssi, 2022-02-23:
Could you update your yaml configuration

```
external_components:
  - source: github://syssi/esphome-atorch-dl24@composite-frames
    refresh: 0s
```

and flash the new version? It should composite short frames now.

nhnam100, 2022-02-24:
new update as your comment and log
[logs_atorch-ble_upload (4).txt](https://github.com/syssi/esphome-atorch-dl24/files/8129220/logs_atorch-ble_upload.4.txt)

syssi, 2022-02-24:
I've made a small fix and added some debug output. Please flash again using this feature branch:

```
external_components:
  - source: github://syssi/esphome-atorch-dl24@composite-frames
    refresh: 0s
```

The `@composite-frames` instead of `@main` is important.

nhnam100, 2022-02-24:
ok, I think it's working  well
[logs_atorch-ble_logs (3).txt](https://github.com/syssi/esphome-atorch-dl24/files/8131655/logs_atorch-ble_logs.3.txt)

commit 5fe40e7c (2022-02-24): Composite notification values (Closes: #10) (#14)
closed 2022-02-24 by syssi