syssi/esphome-atorch-dl24 #51: Sending commands to DL24

TheOriginal92, 2023-05-12:
Hey, is it possible to send commands to a DL24? For example reset measurements, start and stop run, set backlight brightness. Would be very cool.
BR

syssi, 2023-05-13:
It's already supported.

TheOriginal92, 2023-05-13:
> It's already supported.

Can you please lead me the way where i can find the documentation to this function?

syssi, 2023-05-13:
If you use `mqtt` you can control a button if you publish the message "PRESS". The topic depends on
the name of your ESPHome node and the name of the button entity:

```
mosquitto_pub -t 'atorch-usb-meter/button/atorch-usb-meter_reset_energy/command' -m 'PRESS'
mosquitto_pub -t 'atorch-usb-meter/button/atorch-usb-meter_reset_capacity/command' -m 'PRESS'
mosquitto_pub -t 'atorch-usb-meter/button/atorch-usb-meter_reset_runtime/command' -m 'PRESS'
mosquitto_pub -t 'atorch-usb-meter/button/atorch-usb-meter_reset_all/command' -m 'PRESS'
```

If you use Home Assistant the device / esphome node is equipped with some button entities.

https://github.com/syssi/esphome-atorch-dl24/blob/main/esp32-usb-meter-example.yaml#L95-L121

syssi, 2023-05-14:
Does this answer your question and do you know how to proceed?

closed 2023-06-12 by syssi