syssi/esphome-atorch-dl24 #59: DL24M CLC Check Failed

EasilyBoredEngineer, 2023-09-04:
G'day All - 

There's a new kid on the block  the DL24M - see https://www.aliexpress.com/item/1005003426244725.html. 

On mine, the firmware version in the menu is 2.0.

When I use Syssi's contrib, this is the result I get -

`[20:06:09][W][atorch_dl24:209]: CRC check failed (0x47 != 0xAC). Skipping frame
[20:06:09][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.AC (36)
[20:06:10][W][atorch_dl24:209]: CRC check failed (0x47 != 0xCF). Skipping frame
[20:06:10][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.CF (36)
[20:06:11][W][atorch_dl24:209]: CRC check failed (0x47 != 0xEA). Skipping frame
[20:06:11][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.EA (36)
[20:06:12][W][atorch_dl24:209]: CRC check failed (0x47 != 0x8D). Skipping frame
[20:06:12][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.8D (36)
[20:06:13][W][atorch_dl24:209]: CRC check failed (0x47 != 0x28). Skipping frame
[20:06:13][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.28 (36)
[20:06:14][W][atorch_dl24:209]: CRC check failed (0x47 != 0x43). Skipping frame
[20:06:14][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.43 (36)
[20:06:15][W][atorch_dl24:209]: CRC check failed (0x47 != 0x8F). Skipping frame
[20:06:16][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.8F (36)
[20:06:17][W][atorch_dl24:209]: CRC check failed (0x47 != 0x53). Skipping frame
[20:06:17][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.53 (36)
[20:06:17][W][atorch_dl24:209]: CRC check failed (0x47 != 0x9F). Skipping frame
[20:06:17][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.9F (36)
[20:06:19][W][atorch_dl24:209]: CRC check failed (0x47 != 0x63). Skipping frame
[20:06:19][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.63 (36)
[20:06:19][W][atorch_dl24:209]: CRC check failed (0x47 != 0xAF). Skipping frame
[20:06:19][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.AF (36)
[20:06:20][W][atorch_dl24:209]: CRC check failed (0x47 != 0x1D). Skipping frame
[20:06:20][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.1D (36)
[20:06:21][W][atorch_dl24:209]: CRC check failed (0x47 != 0x83). Skipping frame
[20:06:21][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.83 (36)
[20:06:22][W][atorch_dl24:209]: CRC check failed (0x47 != 0x69). Skipping frame
[20:06:22][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.69 (36)
[20:06:23][W][atorch_dl24:209]: CRC check failed (0x47 != 0x57). Skipping frame
[20:06:23][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.57 (36)
[20:06:24][W][atorch_dl24:209]: CRC check failed (0x47 != 0x45). Skipping frame
[20:06:24][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.45 (36)
[20:06:25][W][atorch_dl24:209]: CRC check failed (0x47 != 0x89). Skipping frame
[20:06:25][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.89 (36)
[20:06:26][W][atorch_dl24:209]: CRC check failed (0x47 != 0x55). Skipping frame
[20:06:26][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.55 (36)
[20:06:27][W][atorch_dl24:209]: CRC check failed (0x47 != 0x99). Skipping frame
[20:06:27][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.99 (36)
[20:06:28][W][atorch_dl24:209]: CRC check failed (0x47 != 0x65). Skipping frame
[20:06:28][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.65 (36)
[20:06:29][W][atorch_dl24:209]: CRC check failed (0x47 != 0xC5). Skipping frame
[20:06:29][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.C5 (36)
[20:06:30][W][atorch_dl24:209]: CRC check failed (0x47 != 0xA5). Skipping frame
[20:06:30][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.A5 (36)
[20:06:31][W][atorch_dl24:209]: CRC check failed (0x47 != 0x85). Skipping frame
[20:06:31][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.85 (36)
[20:06:32][W][atorch_dl24:209]: CRC check failed (0x47 != 0xE5). Skipping frame
[20:06:32][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.E5 (36)
[20:06:33][W][atorch_dl24:209]: CRC check failed (0x47 != 0x45). Skipping frame
[20:06:33][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.45 (36)
[20:06:34][W][atorch_dl24:209]: CRC check failed (0x47 != 0x81). Skipping frame
[20:06:34][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.81 (36)
[20:06:35][W][atorch_dl24:209]: CRC check failed (0x47 != 0x45). Skipping frame
[20:06:35][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.45 (36)
[20:06:36][W][atorch_dl24:209]: CRC check failed (0x47 != 0x81). Skipping frame
[20:06:36][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.81 (36)
[20:06:37][W][atorch_dl24:209]: CRC check failed (0x47 != 0x45). Skipping frame
[20:06:37][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.45 (36)
[20:06:38][W][atorch_dl24:209]: CRC check failed (0x47 != 0x81). Skipping frame
[20:06:38][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.81 (36)
[20:06:39][W][atorch_dl24:209]: CRC check failed (0x47 != 0x9A). Skipping frame
[20:06:39][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.9A (36)
[... 141 lines cut ...]
[20:07:46][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.B9 (36)
[20:07:47][W][atorch_dl24:209]: CRC check failed (0x47 != 0x74). Skipping frame
[20:07:47][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.74 (36)
[20:07:48][W][atorch_dl24:209]: CRC check failed (0x47 != 0xAF). Skipping frame
[20:07:48][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.AF (36)
[20:07:49][W][atorch_dl24:209]: CRC check failed (0x47 != 0x62). Skipping frame
[20:07:49][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.62 (36)
[20:07:50][W][atorch_dl24:209]: CRC check failed (0x47 != 0x9D). Skipping frame
[... cut ...]
[20:07:55][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.11 (36)
[20:07:56][W][atorch_dl24:209]: CRC check failed (0x47 != 0xA8). Skipping frame
[20:07:56][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.A8 (36)
[20:07:57][W][atorch_dl24:209]: CRC check failed (0x47 != 0xC7). Skipping frame
[20:07:57][D][atorch_dl24:210]: Payload: FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.C7 (36)`

Any ideas?

syssi, 2023-09-04:
Please disable the CRC check and try again:

```

atorch_dl24:
  - id: atorch0
    ble_client_id: ble_client0
    check_crc: false
    # The meter publishes a status report per second via BLE notification. If you don't like this update interval
    # you can use this setting to throttle the sensor updates by skipping some status reports.
    throttle: 0s
```

It looks like the device is unable to provide a properly calculated checksum:

```
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.AC (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.CF (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.EA (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.8D (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.28 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.43 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.8F (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.53 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.9F (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.63 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.AF (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.1D (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.83 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.69 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.57 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.45 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.89 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.55 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.99 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.65 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.C5 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.A5 (36)
FF.55.01.02.00.00.00.00.00.00.00.00.AD.00.00.00.02.00.00.00.00.00.00.00.00.15.00.00.00.00.3C.00.00.00.00.85 (36)
```

The payload of all frames are equal + a random value at the checksum position.

EasilyBoredEngineer, 2023-09-07:
Thankyou @syssi that worked. For anyone else thinking of buying this version, don't, they're not good. About the most I can draw is 250W without the device shutting down due to overtemp. There's a small heatsink near the DC and sense terminals, with a mosfet. That thing gets stupidly hot - 12V, 48V doesn't matter - not their best product, methinks.

closed 2023-09-07 by EasilyBoredEngineer

syssi, 2025-11-05:
Please go to the `Issues` tab and press the green button at the top right corner to create new issue.

[image]