syssi/esphome-atorch-dl24 #61: DT24PW _

1mwbandy, 2023-10-13:


1mwbandy, 2023-10-13:
Hello Guys,
I need a help. The Bluetooth can not connect to DT24PW.
The board what I use is: ESP-WROOM-32
[image]
The MAC - DT24PW 
[image]

Can you help me?
tnx
Here is the yaml:

substitutions:
  name: atorch-dc-meter
  external_components_source: github://syssi/esphome-atorch-dl24@main
  dl24_mac_address: 38:3B:26:55:76:92
  project_version: 2.0.0
  device_description: "Monitor and control a Atorch meter via bluetooth"

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-atorch-dl24"
    version: ${project_version}

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: Muhely
  password: cirmanska

ota:

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt

api:

esp32_ble_tracker:

ble_client:
  - mac_address: ${dl24_mac_address}
    id: ble_client0

atorch_dl24:
  - id: atorch0
    ble_client_id: ble_client0
    check_crc: false
    # The meter publishes a status report per second via BLE notification. If you don't like this update interval
    # you can use this setting to throttle the sensor updates by skipping some status reports.
    throttle: 1s

binary_sensor:
  - platform: atorch_dl24
    atorch_dl24_id: atorch0
    running:
      name: "${name} running"

sensor:
  - platform: atorch_dl24
[... 132 lines cut ...]
[14:31:08][C][atorch_dl24.button:010]:   Icon: 'mdi:plus-circle-outline'
[14:31:08][C][atorch_dl24.button:010]: AtorchDL24 Button 'atorch-dc-meter minus'
[14:31:08][C][atorch_dl24.button:010]:   Icon: 'mdi:minus-circle-outline'
[14:31:08][C][atorch_dl24.button:010]: AtorchDL24 Button 'atorch-dc-meter setup'
[14:31:08][C][atorch_dl24.button:010]:   Icon: 'mdi:cog'
[14:31:08][C][atorch_dl24.button:010]: AtorchDL24 Button 'atorch-dc-meter enter'
[14:31:08][C][atorch_dl24.button:010]:   Icon: 'mdi:keyboard-return'
[14:31:08][C][esp32_ble:238]: ESP32 BLE:
[14:31:08][C][esp32_ble:240]:   MAC address: B0:A7:32:33:89:CA
[14:31:08][C][esp32_ble:241]:   IO Capability: none
[14:31:08][C][esp32_ble_tracker:617]: BLE Tracker:
[14:31:08][C][esp32_ble_tracker:618]:   Scan Duration: 300 s
[14:31:08][C][esp32_ble_tracker:619]:   Scan Interval: 320.0 ms
[14:31:08][C][esp32_ble_tracker:620]:   Scan Window: 30.0 ms
[14:31:08][C][esp32_ble_tracker:621]:   Scan Type: ACTIVE
[14:31:08][C][esp32_ble_tracker:622]:   Continuous Scanning: True
[14:31:08][C][ble_client:027]: BLE Client:
[14:31:08][C][ble_client:028]:   Address: 38:3B:26:55:76:92
[14:31:08][C][mdns:115]: mDNS:
[14:31:08][C][mdns:116]:   Hostname: atorch-dc-meter
[14:31:08][C][ota:097]: Over-The-Air Updates:
[14:31:08][C][ota:098]:   Address: atorch-dc-meter.local:3232
[14:31:08][C][api:138]: API Server:
[14:31:08][C][api:139]:   Address: atorch-dc-meter.local:6053
[14:31:08][C][api:143]:   Using noise encryption: NO

syssi, 2023-11-04:
Could you try to use the ESP-IDF framework instead of Arduino. Did you change the MAC address to the one of your device?

1mwbandy, 2024-01-16:
Hello,
Right, thank you. With ESP-IDF work fine.

closed 2024-01-16 by syssi