syssi/esphome-atorch-dl24 #66: increase precision on values

quilmore, 2023-12-09:
hi
great code, worked first time!
may I ask to have a way to deal with precision? for example, Voltage in my case is shown as #.## but the measured only changes every 0.1 (like it will show 3.2 for all voltages between 3.20 and 3.29)
for comparison the e_test app on android shows 4 digits of precision, do I need to do something to have it the same way on this? thanks

quilmore, 2023-12-09:
>  4 digits of precision
correction: it's 3 digits in the format ##.###

syssi, 2023-12-09:
Could you provide a screenshot? Is the last digit always zero?

syssi, 2023-12-09:
Could you increase the log level and provide some payload examples of your device?

```
logger:
  baud_rate: 0
  level: VERY_VERBOSE
  logs:
    api.service: WARN
    ota: WARN
    wifi: WARN
    sensor: DEBUG
```

Which model do you own?

quilmore, 2023-12-09:
you're so good
thanks for replying so quick
I've replaced the logger: with that, made no difference to the voltage value, it stays at ##.#0

quilmore, 2023-12-09:
also, it's a DL24

syssi, 2023-12-10:
This cannot be improved in my opinion. The resolution of the voltage measurement at the BLE notification frame is that bad: https://github.com/syssi/esphome-atorch-dl24/blob/main/components/atorch_dl24/atorch_dl24.cpp#L299-L301

If the `e_test app` decodes the same payload I expect the same resolution (probably plus a few zeros). If the app uses a rfcomm connection may be another protocol is used with a different payload / better resolution.

You have increased the log level to `VERY_VERBOSE`. You should see the raw notification data now. Could you provide some of these log messages:

```
      ESP_LOGVV(TAG, "Notification received: %s",
                format_hex_pretty(param->notify.value, param->notify.value_len + 0).c_str());
```

quilmore, 2023-12-10:
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: 62:7B:0F:C6:B5:D9 (RANDOM)
[VV][esp32_ble_tracker:414]:   RSSI: -51
[VV][esp32_ble_tracker:415]:   Name: ''
[VV][esp32_ble_tracker:417]:   TX Power: 2
[VV][esp32_ble_tracker:423]:   Ad Flag: 26
[VV][esp32_ble_tracker:429]:   Manufacturer data: 10.06.3F.1E.3E.25.63.42 (8)
[VV][esp32_ble_tracker:445]: Adv data: 02.01.1A.02.0A.0C.0B.FF.4C.00.10.06.3F.1E.3E.25.63.42 (18)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: 76:5B:B3:F6:06:C0 (RANDOM)
[VV][esp32_ble_tracker:414]:   RSSI: -70
[VV][esp32_ble_tracker:415]:   Name: ''
[VV][esp32_ble_tracker:426]:   Service UUID: 0xFEF3
[VV][esp32_ble_tracker:440]:   Service data:
[VV][esp32_ble_tracker:441]:     UUID: 0xFEF3
[VV][esp32_ble_tracker:442]:     Data: 4A.17.23.50.57.38.55.11.32.B5.44.22.58.C8.24.10.08.13.C0.3A.9E.E7.89.E5.A2.3A.3D (27)
[VV][esp32_ble_tracker:445]: Adv data: 03.03.F3.FE.1E.16.F3.FE.4A.17.23.50.57.38.55.11.32.B5.44.22.58.C8.24.10.08.13.C0.3A.9E.E7.89.E5.A2.3A.3D (35)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: 55:8F:1D:E1:A1:6F (RANDOM)
[VV][esp32_ble_tracker:414]:   RSSI: -90
[VV][esp32_ble_tracker:415]:   Name: ''
[VV][esp32_ble_tracker:423]:   Ad Flag: 26
[VV][esp32_ble_tracker:429]:   Manufacturer data: 09.08.13.02.C0.A8.00.DB.1B.58 (10)
[VV][esp32_ble_tracker:445]: Adv data: 02.01.1A.0D.FF.4C.00.09.08.13.02.C0.A8.00.DB.1B.58 (17)
[V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[V][esp32_ble_client:114]: [0] [FD:34:95:BF:11:7D] gattc_event_handler: event=10 gattc_if=3
[VV][atorch_dl24:142]: Notification received: FF.55.01.02.00.00.22.00.03.82.00.00.56.00.00.00.00.00.00.00.00.00.00.00.00.1C.00.00.38.0E.3C.00.00.00.00.DA (36)
[I][atorch_dl24:253]: Status report received
[D][sensor:094]: 'atorch-dc-meter voltage': Sending state 3.40000 V with 2 decimals of accuracy
[D][sensor:094]: 'atorch-dc-meter current': Sending state 0.89800 A with 3 decimals of accuracy
[D][sensor:094]: 'atorch-dc-meter power': Sending state 3.05320 W with 4 decimals of accuracy
[D][sensor:094]: 'atorch-dc-meter capacity': Sending state 0.86000 Ah with 3 decimals of accuracy
[D][sensor:094]: 'atorch-dc-meter energy': Sending state 0.00000 Wh with 0 decimals of accuracy
[D][sensor:094]: 'atorch-dc-meter temperature': Sending state 28.00000 °C with 0 decimals of accuracy
[D][sensor:094]: 'atorch-dc-meter runtime': Sending state 3374.00000 s with 0 decimals of accuracy
[V][text_sensor:013]: 'atorch-dc-meter runtime formatted': Received new state 56m 14s
[D][text_sensor:064]: 'atorch-dc-meter runtime formatted': Sending state '56m 14s'
[D][sensor:094]: 'atorch-dc-meter dim backlight': Sending state 60.00000 s with 0 decimals of accuracy
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[W][component:214]: Component esp32_ble took a long time for an operation (0.11 s).
[W][component:215]: Components should block for at most 20-30ms.
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: 32:F5:21:40:A1:A5 (RANDOM)
[VV][esp32_ble_tracker:414]:   RSSI: -78
[VV][esp32_ble_tracker:415]:   Name: ''
[VV][esp32_ble_tracker:429]:   Manufacturer data: 01.09.20.22.E5.50.15.D6.12.85.69.C7.65.18.A1.4C.02.83.95.A6.F2.6F.0E.98.B2.97.60 (27)
[VV][esp32_ble_tracker:445]: Adv data: 1E.FF.06.00.01.09.20.22.E5.50.15.D6.12.85.69.C7.65.18.A1.4C.02.83.95.A6.F2.6F.0E.98.B2.97.60 (31)
[V][esp32_ble:178]: (BLE) gap_event_handler - 3
[VV][esp32_ble_tracker:395]: Parse Result:
[VV][esp32_ble_tracker:412]:   Address: 32:F5:21:40:A1:A5 (RANDOM)
[VV][esp32_ble_tracker:414]:   RSSI: -44
[VV][esp32_ble_tracker:415]:   Name: ''
[VV][esp32_ble_tracker:429]:   Manufacturer data: 01.09.20.22.E5.50.15.D6.12.85.69.C7.65.18.A1.4C.02.83.95.A6.F2.6F.0E.98.B2.97.60 (27)
[VV][esp32_ble_tracker:445]: Adv data: 1E.FF.06.00.01.09.20.22.E5.50.15.D6.12.85.69.C7.65.18.A1.4C.02.83.95.A6.F2.6F.0E.98.B2.97.60 (31)

quilmore, 2023-12-10:
power is now displayed with extra precision, thanks for your efforts anyway

closed 2023-12-30 by syssi