syssi/esphome-atorch-dl24 #73: DL24 Ströme in A, Positive & Negative Current Werte

murdock1982, 2024-04-10:
Ist es möglich die Ströme beim entladen den Wert mit einem negativen Vorzeichen zu erhalten? Dies würde Berechnungen ermöglichen.

syssi, 2024-04-10:
You could add a `filter` to multiple the measurement by `-1` to invert the sign:

```
sensor:
  - platform: atorch_dl24
    atorch_dl24_id: atorch0
    voltage:
      name: "${name} voltage"
    current:
      name: "${name} current"
      filters:
        - multiply: -1.0
```

The DL24 isn't able to measure the direction of the flow direction.

murdock1982, 2024-04-11:
The DT24 TW shows on the display whether loading or whether it is unloaded. So he can differentiate in which direction the streams go. Can you also use this information via ESP? Then exactly calculations would be possible.

Thank you very much for the quick answer and a nice day.

syssi, 2024-04-11:
Thanks for the clarification! At the moment the 3 bytes are interpreted as unsigned integer:

https://github.com/syssi/esphome-atorch-dl24/blob/main/docs/protocol-design.md#dc-meter-report
https://github.com/syssi/esphome-atorch-dl24/blob/main/components/atorch_dl24/atorch_dl24.cpp#L304-L306

Could you increase the log level

```
logger:
  baud_rate: 0
  level: VERY_VERBOSE
  logs:
    api.service: WARN
    ota: WARN
    wifi: WARN
    sensor: DEBUG
```

And provide some logs with the RAW traffic (BLE notifications) while pushing current to the device?

syssi, 2024-04-11:
Do you use the ESPHome Dashboard via Home Assistant or how do you update your ESPHome devices? If you use the dashboard the is a `LOG` button per device tile. You should see log messages starting with

```
Notification received: ...
```

here. These are the log messages I'm interested in.

murdock1982, 2024-04-11:
I get the following in the LOGs:

INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/solar01-dc-meter.yaml...
INFO Updating https://github.com/syssi/esphome-atorch-dl24.git@main
INFO UART logging is disabled (baud_rate=0). Not starting UART logs.

syssi, 2024-04-11:
If you read the log using a serial connection instead of over-the-air this line (`baud_rate: 0`) needs to be removed:

```
logger:
  level: VERY_VERBOSE
  logs:
    api.service: WARN
    ota: WARN
    wifi: WARN
    sensor: DEBUG
```

Please try again!

murdock1982, 2024-04-11:
Notification received: FF.55.01.02.00.02.43.00.48.CA.00.27.10.00.00.06.98.00.00.34 (20)

Notification received: 00.00.00.00.00.16.00.04.1F.2C.3C.00.00.00.00.23 (16)

[13:26:54][VV][atorch_dl24:142]: Notification received: FF.55.01.02.00.02.41.00.47.EB.00.27.10.00.00.06.98.00.00.34 (20)
[13:26:54][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[13:26:54][V][esp32_ble_client:119]: [0] [50:54:7B:58:35:95] gattc_event_handler: event=10 gattc_if=3
[13:26:54][VV][atorch_dl24:142]: Notification received: 00.00.00.00.00.16.00.04.1F.2D.3C.00.00.00.00.23 (16)
[13:26:54][I][atorch_dl24:253]: Status report received
[13:26:54][D][sensor:094]: 'solar01-dc-meter voltage': Sending state 57.70000 V with 2 decimals of accuracy
[13:26:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_voltage/state' payload='57.70' retain=1)
[13:26:54][D][sensor:094]: 'solar01-dc-meter current': Sending state 18.41100 A with 3 decimals of accuracy
[13:26:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_current/state' payload='18.411' retain=1)
[13:26:54][D][sensor:094]: 'solar01-dc-meter power': Sending state 1062.31482 W with 4 decimals of accuracy
[13:26:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_power/state' payload='1062.3148' retain=1)
[13:26:54][D][sensor:094]: 'solar01-dc-meter capacity': Sending state 100.00000 Ah with 3 decimals of accuracy
[13:26:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_capacity/state' payload='100.000' retain=1)
[13:26:54][D][sensor:094]: 'solar01-dc-meter energy': Sending state 16880.00000 Wh with 0 decimals of accuracy
[13:26:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_energy/state' payload='16880' retain=1)
[13:26:54][D][sensor:094]: 'solar01-dc-meter temperature': Sending state 22.00000 °C with 0 decimals of accuracy
[13:26:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_temperature/state' payload='22' retain=1)
[13:26:55][D][sensor:094]: 'solar01-dc-meter runtime': Sending state 16305.00000 s with 0 decimals of accuracy
[13:26:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime/state' payload='16305' retain=1)
[13:26:55][V][text_sensor:013]: 'solar01-dc-meter runtime formatted': Received new state 4h 31m 45s
[13:26:55][D][text_sensor:064]: 'solar01-dc-meter runtime formatted': Sending state '4h 31m 45s'
[13:26:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime_formatted/state' payload='4h 31m 45s' retain=1)
[13:26:55][D][sensor:094]: 'solar01-dc-meter dim backlight': Sending state 60.00000 s with 0 decimals of accuracy
[13:26:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_dim_backlight/state' payload='60' retain=1)
[13:26:55][W][component:214]: Component esp32_ble took a long time for an operation (0.07 s).
[13:26:55][W][component:215]: Components should block for at most 20-30ms.

syssi, 2024-04-11:
Perfect! You should see these notifications periodically. Please make sure the display of the DL24 shows a negative current and provide the corresponding raw traffic (notifications). I will verify/decode the data later the day to get an idea the negative sign value is available or not.

murdock1982, 2024-04-11:
Super thank you.
This evening when the sun goes down I will get a negative value on the display and then I will send you the data again.

syssi, 2024-04-11:
Could you tell me the model of your Atorch device?

syssi, 2024-04-11:
```
# 0xFF, 0x55, 0x01, 0x02, 0x00, 0x02, 0x43, 0x00, 0x48, 0xCA, 0x00, 0x27, 0x10, 0x00, 0x00, 0x06, 0x98, 0x00, 0x00, 0x34, 0x00, 0x00, 0x00, 0x00, 0x00, 0x16, 0x00, 0x04, 0x1F, 0x2C, 0x3C, 0x00, 0x00, 0x00, 0x00, 0x23

[13:59:50][I][atorch_dl24:253]: Status report received
[13:59:50][D][sensor:094]: 'atorch-dc-meter voltage': Sending state 57.90000 V with 2 decimals of accuracy
[13:59:50][D][sensor:094]: 'atorch-dc-meter current': Sending state 18.63400 A with 3 decimals of accuracy
[13:59:50][D][sensor:094]: 'atorch-dc-meter power': Sending state 1078.90869 W with 4 decimals of accuracy
[13:59:50][D][sensor:094]: 'atorch-dc-meter capacity': Sending state 100.00000 Ah with 3 decimals of accuracy
[13:59:50][D][sensor:094]: 'atorch-dc-meter energy': Sending state 16880.00000 Wh with 0 decimals of accuracy
[13:59:50][D][sensor:094]: 'atorch-dc-meter temperature': Sending state 22.00000 °C with 0 decimals of accuracy
[13:59:50][D][sensor:094]: 'atorch-dc-meter runtime': Sending state 16304.00000 s with 0 decimals of accuracy
[13:59:50][D][text_sensor:064]: 'atorch-dc-meter runtime formatted': Sending state '4h 31m 44s'
[13:59:50][D][sensor:094]: 'atorch-dc-meter dim backlight': Sending state 60.00000 s with 0 decimals of accuracy
```

murdock1982, 2024-04-11:
DT24TW-300A

murdock1982, 2024-04-11:
[image]

murdock1982, 2024-04-11:
[17:30:54][VV][atorch_dl24:142]: Notification received: 00.00.00.00.00.17.00.08.23.2C.3C.00.00.00.00.23 (16)
[17:30:54][I][atorch_dl24:253]: Status report received
[17:30:54][D][sensor:094]: 'solar01-dc-meter voltage': Sending state 49.80000 V with 2 decimals of accuracy
[17:30:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_voltage/state' payload='49.80' retain=1)
[17:30:54][D][sensor:094]: 'solar01-dc-meter current': Sending state 12.36000 A with 3 decimals of accuracy
[17:30:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_current/state' payload='12.360' retain=1)
[17:30:54][D][sensor:094]: 'solar01-dc-meter power': Sending state 615.52802 W with 4 decimals of accuracy
[17:30:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_power/state' payload='615.5280' retain=1)
[17:30:54][D][sensor:094]: 'solar01-dc-meter capacity': Sending state 80.53000 Ah with 3 decimals of accuracy
[17:30:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_capacity/state' payload='80.530' retain=1)
[17:30:54][D][sensor:094]: 'solar01-dc-meter energy': Sending state 19570.00000 Wh with 0 decimals of accuracy
[17:30:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_energy/state' payload='19570' retain=1)
[17:30:54][D][sensor:094]: 'solar01-dc-meter temperature': Sending state 23.00000 °C with 0 decimals of accuracy
[17:30:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_temperature/state' payload='23' retain=1)
[17:30:54][D][sensor:094]: 'solar01-dc-meter runtime': Sending state 30944.00000 s with 0 decimals of accuracy
[17:30:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime/state' payload='30944' retain=1)
[17:30:54][V][text_sensor:013]: 'solar01-dc-meter runtime formatted': Received new state 8h 35m 44s
[17:30:54][D][text_sensor:064]: 'solar01-dc-meter runtime formatted': Sending state '8h 35m 44s'
[17:30:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime_formatted/state' payload='8h 35m 44s' retain=1)
[17:30:54][D][sensor:094]: 'solar01-dc-meter dim backlight': Sending state 60.00000 s with 0 decimals of accuracy
[17:30:54][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_dim_backlight/state' payload='60' retain=1)
[17:30:54][W][component:214]: Component esp32_ble took a long time for an operation (0.07 s).
[17:30:54][W][component:215]: Components should block for at most 20-30ms.
[17:30:55][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:30:55][V][esp32_ble_client:119]: [0] [50:54:7B:58:35:95] gattc_event_handler: event=10 gattc_if=3
[17:30:55][VV][atorch_dl24:142]: Notification received: FF.55.01.02.00.01.F2.00.2E.E2.00.1F.74.00.00.07.A5.00.00.34 (20)
[17:30:55][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:30:55][V][esp32_ble_client:119]: [0] [50:54:7B:58:35:95] gattc_event_handler: event=10 gattc_if=3
[17:30:55][VV][atorch_dl24:142]: Notification received: 00.00.00.00.00.17.00.08.23.2D.3C.00.00.00.00.23 (16)
[17:30:55][I][atorch_dl24:253]: Status report received
[17:30:55][D][sensor:094]: 'solar01-dc-meter voltage': Sending state 49.80000 V with 2 decimals of accuracy
[17:30:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_voltage/state' payload='49.80' retain=1)
[17:30:55][D][sensor:094]: 'solar01-dc-meter current': Sending state 12.00200 A with 3 decimals of accuracy
[17:30:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_current/state' payload='12.002' retain=1)
[17:30:55][D][sensor:094]: 'solar01-dc-meter power': Sending state 597.69965 W with 4 decimals of accuracy
[17:30:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_power/state' payload='597.6996' retain=1)
[17:30:55][D][sensor:094]: 'solar01-dc-meter capacity': Sending state 80.52000 Ah with 3 decimals of accuracy
[17:30:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_capacity/state' payload='80.520' retain=1)
[17:30:55][D][sensor:094]: 'solar01-dc-meter energy': Sending state 19570.00000 Wh with 0 decimals of accuracy
[17:30:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_energy/state' payload='19570' retain=1)
[17:30:55][D][sensor:094]: 'solar01-dc-meter temperature': Sending state 23.00000 °C with 0 decimals of accuracy
[17:30:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_temperature/state' payload='23' retain=1)
[17:30:55][D][sensor:094]: 'solar01-dc-meter runtime': Sending state 30945.00000 s with 0 decimals of accuracy
[17:30:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime/state' payload='30945' retain=1)
[17:30:55][V][text_sensor:013]: 'solar01-dc-meter runtime formatted': Received new state 8h 35m 45s
[17:30:55][D][text_sensor:064]: 'solar01-dc-meter runtime formatted': Sending state '8h 35m 45s'
[17:30:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime_formatted/state' payload='8h 35m 45s' retain=1)
[17:30:55][D][sensor:094]: 'solar01-dc-meter dim backlight': Sending state 60.00000 s with 0 decimals of accuracy
[17:30:55][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_dim_backlight/state' payload='60' retain=1)
[17:30:55][W][component:214]: Component esp32_ble took a long time for an operation (0.08 s).
[... cut ...]
[17:30:56][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:30:56][V][esp32_ble_client:119]: [0] [50:54:7B:58:35:95] gattc_event_handler: event=10 gattc_if=3
[17:30:56][VV][atorch_dl24:142]: Notification received: FF.55.01.02.00.01.F2.00.2E.D6.00.1F.74.00.00.07.A6.00.00.34 (20)
[17:30:56][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:30:56][V][esp32_ble_client:119]: [0] [50:54:7B:58:35:95] gattc_event_handler: event=10 gattc_if=3
[17:30:56][VV][atorch_dl24:142]: Notification received: 00.00.00.00.00.17

murdock1982, 2024-04-11:
[image]

murdock1982, 2024-04-11:
[17:30:55][I][atorch_dl24:253]: Status report received
[17:30:55][D][sensor:094]: 'solar01-dc-meter voltage': Sending state 49.80000 V with 2 decimals of accuracy

[17:30:55][D][sensor:094]: 'solar01-dc-meter current': Sending state 12.00200 A with 3 decimals of accuracy

[17:30:55][D][sensor:094]: 'solar01-dc-meter power': Sending state 597.69965 W with 4 decimals of accuracy

[17:30:55][D][sensor:094]: 'solar01-dc-meter capacity': Sending state 80.52000 Ah with 3 decimals of accuracy

[17:30:55][D][sensor:094]: 'solar01-dc-meter energy': Sending state 19570.00000 Wh with 0 decimals of accuracy

[17:30:55][D][sensor:094]: 'solar01-dc-meter temperature': Sending state 23.00000 °C with 0 decimals of accuracy

[17:30:55][D][sensor:094]: 'solar01-dc-meter runtime': Sending state 30945.00000 s with 0 decimals of accuracy

murdock1982, 2024-04-11:
[image]

syssi, 2024-04-11:
Could you provide some more frames / log lines prefixed with `Notification received`?

syssi, 2024-04-11:
```
     0  FF FF
     1  55 55
     2  01 01
     3  02 02 Device type
     4  00 00 Voltage
     5  02 01 Voltage
     6  43 F2 Voltage     579=57.9V, 498=49.8V
     7  00 00 Current
     8  48 2E Current
     9  CA E2 Current     18634=18.634A, 12002=12.002A
    10  00 00 Capacity
    11  27 1F Capacity
    12  10 74 Capacity    10000=100Ah, 8052=80.52Ah
    13  00 00 Energy
    14  00 00 Energy
    15  06 07 Energy
    16  98 A5 Energy      1688*10=16880Wh,  1957*10=19570Wh
    17  00 00 Price?
    18  00 00 Price?
    19  34 34 Price?
    20  00 00 Unknown
    21  00 00 Unknown
    22  00 00 Unknown
    23  00 00 Unknown
    24  00 00 Temperature
    25  16 17 Temperature
    26  00 00 Hour
    27  04 08 Hour
    28  1F 23 Minute
    29  2C 2D Second
    30  3C 3C Backlight
    31  00 00
    32  00 00
    33  00 00
    34  00 00
    35  23 23 CRC
```

It looks like the BLE notifications doesn't include any details about the direction of the current flow. :-(

murdock1982, 2024-04-11:
[18:53:01][I][atorch_dl24:253]: Status report received
[18:53:01][D][sensor:094]: 'solar01-dc-meter voltage': Sending state 49.20000 V with 2 decimals of accuracy
[18:53:01][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_voltage/state' payload='49.20' retain=1)
[18:53:01][D][sensor:094]: 'solar01-dc-meter current': Sending state 20.32200 A with 3 decimals of accuracy
[18:53:01][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_current/state' payload='20.322' retain=1)
[18:53:01][D][sensor:094]: 'solar01-dc-meter power': Sending state 999.84247 W with 4 decimals of accuracy
[18:53:01][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_power/state' payload='999.8425' retain=1)
[18:53:01][D][sensor:094]: 'solar01-dc-meter capacity': Sending state 66.68000 Ah with 3 decimals of accuracy
[18:53:01][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_capacity/state' payload='66.680' retain=1)
[18:53:01][D][sensor:094]: 'solar01-dc-meter energy': Sending state 20290.00000 Wh with 0 decimals of accuracy
[18:53:01][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_energy/state' payload='20290' retain=1)
[18:53:01][D][sensor:094]: 'solar01-dc-meter temperature': Sending state 22.00000 °C with 0 decimals of accuracy
[18:53:01][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_temperature/state' payload='22' retain=1)
[18:53:01][D][sensor:094]: 'solar01-dc-meter runtime': Sending state 35871.00000 s with 0 decimals of accuracy
[18:53:01][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime/state' payload='35871' retain=1)
[18:53:01][V][text_sensor:013]: 'solar01-dc-meter runtime formatted': Received new state 9h 57m 51s
[18:53:01][D][text_sensor:064]: 'solar01-dc-meter runtime formatted': Sending state '9h 57m 51s'
[18:53:01][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime_formatted/state' payload='9h 57m 51s' retain=1)
[18:53:01][D][sensor:094]: 'solar01-dc-meter dim backlight': Sending state 60.00000 s with 0 decimals of accuracy
[18:53:01][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_dim_backlight/state' payload='60' retain=1)
[18:53:01][W][component:214]: Component esp32_ble took a long time for an operation (0.08 s).
[18:53:01][W][component:215]: Components should block for at most 20-30ms.
[18:53:01][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[18:53:01][V][esp32_ble_client:119]: [0] [50:54:7B:58:35:95] gattc_event_handler: event=10 gattc_if=3
[18:53:01][VV][atorch_dl24:142]: Notification received: FF.55.01.02.00.01.EC.00.4E.B9.00.1A.0B.00.00.07.ED.00.00.34 (20)
[18:53:01][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[18:53:01][V][esp32_ble_client:119]: [0] [50:54:7B:58:35:95] gattc_event_handler: event=10 gattc_if=3
[18:53:01][VV][atorch_dl24:142]: Notification received: 00.00.00.00.00.16.00.09.39.34.3C.00.00.00.00.23 (16)
[18:53:02][I][atorch_dl24:253]: Status report received
[18:53:02][D][sensor:094]: 'solar01-dc-meter voltage': Sending state 49.20000 V with 2 decimals of accuracy
[18:53:02][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_voltage/state' payload='49.20' retain=1)
[18:53:02][D][sensor:094]: 'solar01-dc-meter current': Sending state 20.15300 A with 3 decimals of accuracy
[18:53:02][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_current/state' payload='20.153' retain=1)
[18:53:02][D][sensor:094]: 'solar01-dc-meter power': Sending state 991.52771 W with 4 decimals of accuracy
[18:53:02][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_power/state' payload='991.5277' retain=1)
[18:53:02][D][sensor:094]: 'solar01-dc-meter capacity': Sending state 66.67000 Ah with 3 decimals of accuracy
[18:53:02][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_capacity/state' payload='66.670' retain=1)
[18:53:02][D][sensor:094]: 'solar01-dc-meter energy': Sending state 20290.00000 Wh with 0 decimals of accuracy
[18:53:02][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_energy/state' payload='20290' retain=1)
[18:53:02][D][sensor:094]: 'solar01-dc-meter temperature': Sending state 22.00000 °C with 0 decimals of accuracy
[18:53:02][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_temperature/state' payload='22' retain=1)
[18:53:02][D][sensor:094]: 'solar01-dc-meter runtime': Sending state 35872.00000 s with 0 decimals of accuracy
[18:53:02][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime/state' payload='35872' retain=1)
[18:53:02][V][text_sensor:013]: 'solar01-dc-meter runtime formatted': Received new state 9h 57m 52s
[18:53:02][D][text_sensor:064]: 'solar01-dc-meter runtime formatted': Sending state '9h 57m 52s'
[18:53:02][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_runtime_formatted/state' payload='9h 57m 52s' retain=1)
[18:53:02][D][sensor:094]: 'solar01-dc-meter dim backlight': Sending state 60.00000 s with 0 decimals of accuracy
[18:53:02][V][mqtt:474]: Publish(topic='solar01-dc-meter/sensor/solar01-dc-meter_dim_backlight/state' payload='60' retain=1)
[18:53:02][W][component:214]: Component esp32_ble took a long time for an operation (0.08 s).
[18:53:02][W][component:215]: Components should block for at most 20-30ms.
[18:53:02][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[... cut ...]
[18:53:06][V][esp32_ble_client:119]: [0] [50:54:7B:58:35:95] gattc_event_handler: event=10 gattc_if=3
[18:53:06][VV][atorch_dl24:142]: Notification received: FF.55.01.02.00.01.EC.00.50.46.00.1A.09.00.00.07.ED.00.00.34 (20)
[18:53:06][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[18:53:06][V][esp32_ble_client:119]: [0] [50:54:7B:58:35:95] gattc_event_handler: event=10 gattc_if=3
[18:53:06][VV][atorch_dl24:142]: Notification received: 00.00.00.00.00.16.00.09.39.38.3C.00.00.00.00.23 (16)
[18:53:06][I][atorch_dl24:253]: Status report received

murdock1982, 2024-04-11:
I'll try the Windows tool to see if it sends the data. But I can't do it until I'm home.

syssi, 2024-04-11:
Sounds good. Please try the OEM apps (windows vs. android) and have a look if these apps are able to show the direction.

murdock1982, 2024-04-11:
[image]

murdock1982, 2024-04-11:
Discharge

syssi, 2024-04-11:
I don't understand. ;-) What's the result of your analysis?

murdock1982, 2024-04-11:
Moments the batteries will be discharged. But the value of electricity is positive. When the batteries are charging, it is also positive. So the display shows positives and negatives, but nothing is sent via Bluetooth.

Hmm

Do you know an alternative, other shunt and display etc?

syssi, 2024-04-12:
Unfortunately, I have no recommendations.

Let's close this issue because there is no room for improvements at the moment.

closed 2024-04-12 by syssi

murdock1982, 2024-04-12:
Okay