syssi/esphome-atorch-dl24 #76: No BLE connection on DL24M Purple

Viper7000, 2024-06-11:
I have such a device https://www.aliexpress.com/item/1005004042609941.html
Firmware v 2.1
I have ESP32-CAM - that's all I have in stock
I use   framework: type: esp-idf. And I not see ble_client: connected. Instead I only see in the logs "[V][esp32_ble:314]: (BLE) gap_event_handler - 3"

```
[18:00:37][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:37][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:38][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:38][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:39][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:39][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:39][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:39][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:40][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:40][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:41][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:41][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[18:00:41][V][esp32_ble:314]: (BLE) gap_event_handler - 3
```

I can connect from PC and from mobile.

what could be wrong?

syssi, 2024-06-11:
Please increase the log level and provide a complete log:

```
logger:
  level: VERY_VERBOSE
  logs:
    ota: INFO
    wifi: INFO
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

Viper7000, 2024-06-11:
I'll definitely do it tomorrow. The power just went out throughout the city.
There is nothing interesting in the logs other than what I wrote above :)
Sees other BLEs, but there is no DL24M among them

syssi, 2024-06-11:
I'm interested of the BLE advertisement of your DL24M. May be the service UUIDs / BLE module has changed slightly.

Viper7000, 2024-06-11:
My config
```
substitutions:
  name: DL24M
  external_components_source: github://syssi/esphome-atorch-dl24@main
  dl24_mac_address: 4B:34:33:EB:92:8D
  project_version: 2.0.0
  device_description: "Monitor and control a Atorch meter via bluetooth"

esphome:
  name: atorch-dl24m
  friendly_name: Atorch-DL24M
  comment: ${device_description}
  project:
    name: "syssi.esphome-atorch-dl24"
    version: ${project_version}

esp32:
  board: esp32dev
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

# Enable logging
logger:
  level: DEBUG
    
debug:
  update_interval: 5s

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  - ssid: !secret wifi_ssid2
    password: !secret wifi_password2

captive_portal:

esp32_ble_tracker:

ble_client:
  - mac_address: ${dl24_mac_address}
    id: ble_client0

atorch_dl24:
  - id: atorch0
    ble_client_id: ble_client0
    check_crc: false
    # The meter publishes a status report per second via BLE notification. If you don't like this update interval
    # you can use this setting to throttle the sensor updates by skipping some status reports.
    throttle: 0s

binary_sensor:
  - platform: atorch_dl24
    atorch_dl24_id: atorch0
    running:
      name: "${name} running"

sensor:
  - platform: uptime
    name: ESP32 Uptime
    filters:
      - lambda: return x / 3600.0;
    unit_of_measurement: h
    accuracy_decimals: 1
  - platform: wifi_signal
    name: ESP32 RSSI
    update_interval: 300s  
  - platform: esp32_hall
    name: "ESP32 Hall Sensor"
    update_interval: 300s
    entity_category: "diagnostic"
  - platform: debug
    free:
      name: "Heap Free"
    loop_time:
      name: "Loop Time"

  - platform: atorch_dl24
    atorch_dl24_id: atorch0
    voltage:
      name: "${name} voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    capacity:
      name: "${name} capacity"
    energy:
      name: "${name} energy"
    temperature:
      name: "${name} temperature"
    dim_backlight:
      name: "${name} dim backlight"
    runtime:
      name: "${name} runtime"

text_sensor:
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"
  - platform: wifi_info
    ip_address:
      name: ESP32 IP Address
    ssid:
      name: ESP32 SSID

  - platform: atorch_dl24
    atorch_dl24_id: atorch0
    runtime_formatted:
      name: "${name} runtime formatted"

button:
  - platform: atorch_dl24
    atorch_dl24_id: atorch0
    reset_energy:
      name: "${name} reset energy"
    reset_capacity:
      name: "${name} reset capacity"
    reset_runtime:
      name: "${name} reset runtime"
    reset_all:
      name: "${name} reset all"
    plus:
      name: "${name} plus"
    minus:
      name: "${name} minus"
    setup:
      name: "${name} setup"
    enter:
      name: "${name} enter"
```

syssi, 2024-06-11:
Please replace

```
logger:
  level: DEBUG
```

with

```
logger:
  level: VERY_VERBOSE
  logs:
    ota: INFO
    wifi: INFO
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

Viper7000, 2024-06-11:
Is that enough? :)
BLE Scan I tried different settings both active and passive
```
[22:45:33][I][app:100]: ESPHome version 2024.5.3 compiled on Jun 11 2024, 22:43:31
[22:45:33][I][app:102]: Project syssi.esphome-atorch-dl24 version 2.0.0
[22:45:33][C][logger:185]: Logger:
[22:45:33][C][logger:186]:   Level: VERY_VERBOSE
[22:45:33][C][logger:188]:   Log Baud Rate: 115200
[22:45:33][C][logger:189]:   Hardware UART: UART0
[22:45:33][C][logger:193]:   Level for 'ota': INFO
[22:45:33][C][logger:193]:   Level for 'wifi': INFO
[22:45:33][C][logger:193]:   Level for 'scheduler': DEBUG
[22:45:33][C][logger:193]:   Level for 'component': DEBUG
[22:45:33][C][logger:193]:   Level for 'sensor': DEBUG
[22:45:33][C][logger:193]:   Level for 'mqtt': INFO
[22:45:33][C][logger:193]:   Level for 'mqtt.idf': INFO
[22:45:33][C][logger:193]:   Level for 'mqtt.component': INFO
[22:45:33][C][logger:193]:   Level for 'mqtt.sensor': INFO
[22:45:33][C][logger:193]:   Level for 'mqtt.switch': INFO
[22:45:33][C][logger:193]:   Level for 'api.service': INFO
[22:45:33][C][logger:193]:   Level for 'api': INFO
[22:45:33][C][uptime.sensor:031]: Uptime Sensor 'ESP32 Uptime'
[22:45:33][C][uptime.sensor:031]:   Device Class: 'duration'
[22:45:33][C][uptime.sensor:031]:   State Class: 'total_increasing'
[22:45:33][C][uptime.sensor:031]:   Unit of Measurement: 'h'
[22:45:33][C][uptime.sensor:031]:   Accuracy Decimals: 1
[22:45:33][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[22:45:33][V][uptime.sensor:031]:   Unique ID: 'a842e3566704-uptime'
[22:45:33][C][atorch_dl24:065]: DL24
[22:45:33][C][atorch_dl24:066]:  Voltage 'DL24M voltage'
[22:45:33][C][atorch_dl24:066]:    Device Class: 'voltage'
[22:45:33][C][atorch_dl24:066]:    State Class: 'measurement'
[22:45:33][C][atorch_dl24:066]:    Unit of Measurement: 'V'
[22:45:33][C][atorch_dl24:066]:    Accuracy Decimals: 2
[22:45:33][C][atorch_dl24:067]:  Current 'DL24M current'
[22:45:33][C][atorch_dl24:067]:    Device Class: 'current'
[22:45:33][C][atorch_dl24:067]:    State Class: 'measurement'
[22:45:33][C][atorch_dl24:067]:    Unit of Measurement: 'A'
[22:45:33][C][atorch_dl24:067]:    Accuracy Decimals: 3
[22:45:33][C][atorch_dl24:068]:  Power 'DL24M power'
[22:45:33][C][atorch_dl24:068]:    State Class: 'measurement'
[22:45:33][C][atorch_dl24:068]:    Unit of Measurement: 'W'
[22:45:33][C][atorch_dl24:068]:    Accuracy Decimals: 4
[22:45:33][C][atorch_dl24:069]:  Capacity 'DL24M capacity'
[22:45:33][C][atorch_dl24:069]:    Device Class: 'energy'
[22:45:33][C][atorch_dl24:069]:    State Class: 'total_increasing'
[22:45:33][C][atorch_dl24:069]:    Unit of Measurement: 'Ah'
[22:45:33][C][atorch_dl24:069]:    Accuracy Decimals: 3
[22:45:33][C][atorch_dl24:069]:    Icon: 'mdi:battery-medium'
[22:45:33][C][atorch_dl24:070]:  Energy 'DL24M energy'
[22:45:33][C][atorch_dl24:070]:    Device Class: 'energy'
[22:45:33][C][atorch_dl24:070]:    State Class: 'total_increasing'
[22:45:33][C][atorch_dl24:070]:    Unit of Measurement: 'Wh'
[22:45:33][C][atorch_dl24:070]:    Accuracy Decimals: 0
[22:45:33][C][atorch_dl24:070]:    Icon: 'mdi:counter'
[22:45:33][C][atorch_dl24:071]:  Temperature 'DL24M temperature'
[22:45:33][C][atorch_dl24:071]:    Device Class: 'temperature'
[22:45:33][C][atorch_dl24:071]:    State Class: 'measurement'
[22:45:33][C][atorch_dl24:071]:    Unit of Measurement: '°C'
[22:45:33][C][atorch_dl24:071]:    Accuracy Decimals: 0
[22:45:33][C][atorch_dl24:072]:  Dim Backlight 'DL24M dim backlight'
[22:45:33][C][atorch_dl24:072]:    State Class: 'measurement'
[22:45:33][C][atorch_dl24:072]:    Unit of Measurement: 's'
[22:45:33][C][atorch_dl24:072]:    Accuracy Decimals: 0
[22:45:33][C][atorch_dl24:072]:    Icon: 'mdi:timer-outline'
[22:45:33][C][atorch_dl24:073]:  Running 'DL24M running'
[22:45:33][C][atorch_dl24:079]:  Runtime 'DL24M runtime'
[22:45:33][C][atorch_dl24:079]:    State Class: 'total_increasing'
[22:45:33][C][atorch_dl24:079]:    Unit of Measurement: 's'
[22:45:33][C][atorch_dl24:079]:    Accuracy Decimals: 0
[... 258 lines cut ...]
[22:45:39][VV][esp32_ble_tracker:444]:   Address: A4:C1:38:39:CE:A4 (PUBLIC)
[22:45:39][VV][esp32_ble_tracker:446]:   RSSI: -90
[22:45:39][VV][esp32_ble_tracker:447]:   Name: ''
[22:45:39][VV][esp32_ble_tracker:455]:   Ad Flag: 6
[22:45:39][VV][esp32_ble_tracker:472]:   Service data:
[22:45:39][VV][esp32_ble_tracker:473]:     UUID: 0xFE95
[22:45:39][VV][esp32_ble_tracker:474]:     Data: 30.58.5B.05.03.A4.CE.39.38.C1.A4.08 (12)
[22:45:39][VV][esp32_ble_tracker:477]: Adv data: 02.01.06.0F.16.95.FE.30.58.5B.05.03.A4.CE.39.38.C1.A4.08 (19)
[22:45:40][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[22:45:40][VV][esp32_ble_tracker:427]: Parse Result:
[22:45:40][VV][esp32_ble_tracker:444]:   Address: 7C:C2:94:17:68:DE (PUBLIC)
[22:45:40][VV][esp32_ble_tracker:446]:   RSSI: -59
[22:45:40][VV][esp32_ble_tracker:447]:   Name: ''
[22:45:40][VV][esp32_ble_tracker:455]:   Ad Flag: 6
[22:45:40][VV][esp32_ble_tracker:461]:   Manufacturer data: 0A.10.93.09.00.DC.68.17.94.C2.7C.81 (12)
[22:45:40][VV][esp32_ble_tracker:477]: Adv data: 02.01.06.0F.FF.8F.03.0A.10.93.09.00.DC.68.17.94.C2.7C.81 (19)
[22:45:40][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[22:45:40][VV][esp32_ble_tracker:427]: Parse Result:
[22:45:40][VV][esp32_ble_tracker:444]:   Address: 7C:C2:94:17:68:DE (PUBLIC)
[22:45:40][VV][esp32_ble_tracker:446]:   RSSI: -59
[22:45:40][VV][esp32_ble_tracker:447]:   Name: ''
[22:45:40][VV][esp32_ble_tracker:455]:   Ad Flag: 6
[22:45:40][VV][esp32_ble_tracker:461]:   Manufacturer data: 0A.10.93.09.00.DC.68.17.94.C2.7C.81 (12)
[22:45:40][VV][esp32_ble_tracker:477]: Adv data: 02.01.06.0F.FF.8F.03.0A.10.93.09.00.DC.68.17.94.C2.7C.81 (19)
```

Viper7000, 2024-06-11:
here is my device in the logs
```
[22:45:39][V][esp32_ble:314]: (BLE) gap_event_handler - 3
[22:45:39][VV][esp32_ble_tracker:427]: Parse Result:
[22:45:39][VV][esp32_ble_tracker:444]:   Address: 4A:34:33:EB:92:8D (PUBLIC)
[22:45:39][VV][esp32_ble_tracker:446]:   RSSI: -51
[22:45:39][VV][esp32_ble_tracker:447]:   Name: ''
[22:45:39][VV][esp32_ble_tracker:455]:   Ad Flag: 6
[22:45:39][VV][esp32_ble_tracker:458]:   Service UUID: 0xFFE0
[22:45:39][VV][esp32_ble_tracker:477]: Adv data: 02.01.06.03.02.E0.FF (7)
```

Viper7000, 2024-06-11:
And my device have 2 different connections on BLE:  DL24M_SPP and DL24M_BLE
And they have different MAC! I tried to connect to DL24M_SPP. This is right?

Viper7000, 2024-06-11:
I was able to start the device. Need to connect to DL24M_BLE. Not to DL24M_SPP.
But on PC I connect to DL24M_SPP.

Working all buttons except "plus" and "minus". 
The device responds to all presses with sound. Pressing plus and minus - no sound.
This is log
```
[00:21:44][D][button:010]: 'DL24M plus' Pressed.
[00:21:44][I][atorch_dl24:053]: Write command: FF.55.11.02.11.00.00.00.00.60 (10)
[00:21:44][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[00:21:44][V][esp32_ble_client:121]: [0] [4A:34:33:EB:92:8D] gattc_event_handler: event=4 gattc_if=3
[00:21:44][D][esp32_ble_client:110]: [0] [4A:34:33:EB:92:8D] ESP_GATTC_WRITE_CHAR_EVT
[00:21:45][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[00:21:45][V][esp32_ble_client:121]: [0] [4A:34:33:EB:92:8D] gattc_event_handler: event=10 gattc_if=3
[00:21:45][D][esp32_ble_client:110]: [0] [4A:34:33:EB:92:8D] ESP_GATTC_NOTIFY_EVT
[00:21:45][VV][atorch_dl24:142]: Notification received: FF.55.01.02.00.00.8F.00.00.00.00.00.07.00.00.00.00.00.00.00.00.00.00.00.00.1A.00.00.00.00.3C.00.00.00.00.01 (36)
[00:21:45][I][atorch_dl24:253]: Status report received

[00:21:43][D][button:010]: 'DL24M minus' Pressed.
[00:21:43][I][atorch_dl24:053]: Write command: FF.55.11.02.12.00.00.00.00.61 (10)
[00:21:43][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[00:21:43][V][esp32_ble_client:121]: [0] [4A:34:33:EB:92:8D] gattc_event_handler: event=4 gattc_if=3
[00:21:43][D][esp32_ble_client:110]: [0] [4A:34:33:EB:92:8D] ESP_GATTC_WRITE_CHAR_EVT
[00:21:43][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[00:21:43][V][esp32_ble_client:121]: [0] [4A:34:33:EB:92:8D] gattc_event_handler: event=10 gattc_if=3
[00:21:43][D][esp32_ble_client:110]: [0] [4A:34:33:EB:92:8D] ESP_GATTC_NOTIFY_EVT
[00:21:43][VV][atorch_dl24:142]: Notification received: FF.55.01.02.00.00.8F.00.00.00.00.00.07.00.00.00.00.00.00.00.00.00.00.00.00.1A.00.00.00.00.3C.00.00.00.00.51 (36)
[00:21:43][I][atorch_dl24:253]: Status report received
```
Pres "setup" working and device responds with sound.
This is log
```
[00:25:22][D][button:010]: 'DL24M setup' Pressed.
[00:25:22][I][atorch_dl24:053]: Write command: FF.55.11.02.31.00.00.00.00.00 (10)
[00:25:22][V][esp32_ble:342]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[00:25:22][V][esp32_ble_client:121]: [0] [4A:34:33:EB:92:8D] gattc_event_handler: event=10 gattc_if=3
[00:25:22][D][esp32_ble_client:110]: [0] [4A:34:33:EB:92:8D] ESP_GATTC_NOTIFY_EVT
[00:25:22][VV][atorch_dl24:142]: Notification received: FF.55.01.02.00.00.8F.00.00.00.00.00.07.00.00.00.00.00.00.00.00.00.00.00.00.1A.00.00.00.00.3C.00.00.00.00.77 (36)
[00:25:22][I][atorch_dl24:253]: Status report received
```

Viper7000, 2024-06-11:
The readings do not match the device:

DL24         -   ESP         -   PC (DL24_SPP)
14.38V      -  14.30V     -  14.38V
70.32W     -  70.00W    -  70.32W
2.187Wh    -  0Wh        -  2.187Wh
155mAh    -  0.150Ah  -  155mAh
26.70°C     -  26°C        -   26.70°C

syssi, 2024-06-11:
Good job! I will have a look what needs to be improved to support your model properly. The raw traffic and your notes are pretty helpful!

Viper7000, 2024-06-11:
I analyzed the raw traffic and it looks like the device is not transmitting complete data :(
There is no accuracy on ESP that I see on the DL screen, but on the PC everything is accurate

syssi, 2024-06-12:
Correct. The resolution of the BLE traffic isn't great.

Viper7000, 2024-06-12:
But why is the accuracy on PC the same as on DL?
Can I use DL24M_SPP on ESP?

syssi, 2024-06-12:
Mobile and PC uses the SPP (Serial Port Profile) connection. SPP isn't supported by ESPHome and this project only supports the BLE protocol.

closed 2024-06-12 by syssi

Viper7000, 2024-06-12:
I check now. Mobile app use DL24M_BLE and same accuracy as ESP :(

Viper7000, 2024-06-12:
But from mobile work plus and minus buttons :)

syssi, 2024-06-12:
Good to know! :-) You could capture the BLE traffic using btsnoop:

https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa

Could you provide such a capture including some +/- presses?

Viper7000, 2024-06-12:
That's what you need?

Plus    - FF 55 11 02 33 00 00 00 00 02
Minus - FF 55 11 02 34 00 00 00 00 03

syssi, 2024-06-12:
Perfect!

If we compare these payloads with the implemented ones there is a difference at byte 4:

https://github.com/syssi/esphome-atorch-dl24/blob/main/components/atorch_dl24/button/__init__.py#L12-L21

``` 
FF.55.11.01.11.00.00.00.00.67 plus
FF.55.11.01.12.00.00.00.00.60 minus
FF.55.11.03.33.00.00.00.00.03 usb_plus
FF.55.11.03.34.00.00.00.00.0c usb_minus

vs.

FF.55.11.02.33.00.00.00.00.02 dl24m_plus
FF.55.11.02.34.00.00.00.00.03 dl24m_minus
```