syssi/esphome-atorch-dl24 #82: Calculate Energy on the fly?

makerwolf1, 2024-08-25:
Hey! Thanks for your great work, it adds massive value to the electronic load! 

The DL24 seems to only report energy in 10wH steps, despite displaying it with 3 digits accuracy on the display. Since we have both voltage and current it would be trivial to calculate energy on the fly, but unfortunately there is no way to add a `on_value:` hook to the sensors to trigger update of a template sensor. 

Any chance to maybe add that as an additional sensor to the platform?

syssi, 2024-08-25:
The `on_value` automations are supported by every sensor entity. Please try something like this:

```
sensor:
  - platform: template
    id: power_computed
    name: "${name} power computed"
    update_interval: never

  - platform: atorch_dl24
    atorch_dl24_id: atorch0
    voltage:
      name: "${name} voltage"
      id: voltage
    current:
      name: "${name} current"
      id: current
      on_value:
        then:
          - sensor.template.publish:
              id: power_computed
              state: !lambda 'return id(voltage).state * id(current).state;'
```

makerwolf1, 2024-08-25:
ahh! that's what threw me off, of course it is inside the sensors and not the platform. thank you so much for the swift reply. 

how would you recommend going about the timestamp? I just realized that it's actually difficult getting a reliable reading in case packets are missed. it's not quite as straight forward as calling millis() and store the difference, although that would probably still yield more accuracy than 10wH increments ...

syssi, 2024-08-25:
You have to feed the `total_daily_energy` with the `power` measurement: https://esphome.io/components/sensor/total_daily_energy.html

I would not attempt to calculate it myself.

makerwolf1, 2024-08-25:
Gotta love the simplicity and matureness of esphome. thank you syssi! I actually went with the integration sensor because I don't want it to be reset at midnight.

thank you!

closed 2024-08-25 by makerwolf1

syssi, 2024-08-25:
Have fun! :-)