syssi/esphome-atorch-dl24 #85: voltage value is rounded

man55, 2024-09-12:
If you look at the device screen, the voltage is 3.352v and varies slightly. 
But through integration into the HA I see only 3.3v, which is a too rounded value, and this makes the integration essentially useless since I need to track the discharge of the LFP cell.

I don't know what exactly the problem is.
Is there any way to fix this?

[image]
[image]

p.s. Another question: is it possible to add external temperature to the integration?

syssi, 2024-09-12:
Does this feedback help: https://github.com/syssi/esphome-atorch-dl24/discussions/84

syssi, 2024-09-12:
> p.s. Another question: is it possible to add external temperature to the integration?

You can add any (ESPHome supported) temperature sensor to the YAML.

syssi, 2024-09-12:
> But through integration into the HA I see only 3.3v, which is a too rounded value, and this makes the integration essentially useless since I need to track the discharge of the LFP cell.

The payload of the bluetooth notification frame doesn't provide a better resolution. See https://github.com/syssi/esphome-atorch-dl24/blob/main/docs/protocol-design.md#dc-meter-report

closed 2024-09-12 by syssi