syssi/esphome-atorch-dl24 #91: Energy sensor not working

hapklaar, 2025-02-11:
For me the sensors Energy and runtime stay empty for my DL24. I used the sensor definition from you dc example

```
sensor:
  - platform: atorch_dl24
    atorch_dl24_id: atorch0
    voltage:
      name: "${name} voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    capacity:
      name: "${name} capacity"
    energy:
      name: "${name} energy"
    temperature:
      name: "${name} temperature"
    dim_backlight:
      name: "${name} dim backlight"
    runtime:
      name: "${name} runtime"
```

Have you seen this before?

[image]

syssi, 2025-02-11:
Could you increase the log level and provide a raw frame in plaintext? It's printed to the log.

```
logger:
  level: VERY_VERBOSE
  logs:
    ota: INFO
    wifi: INFO
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

syssi, 2025-02-12:
Could you provide a photo of your DL24?

hapklaar, 2025-02-12:
Sure, here you go

[image]

[image]