syssi/esphome-atorch-dl24 #94: not work after reboot ESP32

polzovatel8, 2025-03-16:
Hello. I have such a problem. I have an AT3010 energy meter, as soon as I upload the firmware to ESP32 everything works, the data goes, I attach a log. After rebooting ESP32, the data does not go. Help.

[logs_at3010_work.txt](https://github.com/user-attachments/files/19274878/logs_at3010_work.txt)

[logs_at3010_after_reboot_no_work.txt](https://github.com/user-attachments/files/19274879/logs_at3010_after_reboot_no_work.txt)

syssi, 2025-03-16:
How do you do the reboot? Per software or by removing the power?

polzovatel8, 2025-03-16:
Reboot after power disconnected

syssi, 2025-03-16:
Does it work again if you reboot the dummy load too?

polzovatel8, 2025-03-16:
no matter how many power reboots, it doesn't work. only downloading the firmware gives some result of work until the first reboot

syssi, 2025-03-17:
Could you try to use another ESP?

polzovatel8, 2025-03-17:
Unfortunately I don't have another one, I ordered it and will try it when it arrives.

syssi, 2025-03-17:
Could you tell me the ESP32 model you are using at the moment?

polzovatel8, 2025-03-17:
Esp32-wroom-32D

polzovatel8, 2025-03-17:
[image]

polzovatel8, 2025-03-29:
HI. i buy new esp32. flash firmware, and not work again.

closed 2025-06-29 by syssi